LilyGO/TTGO-LoRa32-V2.1 #8: TTGO-LoRa32-V2.1 T3_V1.6 timeout uploading code / access GPIO0?

umbm, 2020-12-06:
My new board is exactly as shown here https://github.com/LilyGO/TTGO-LoRa32-V2.1
How to access GPIO0 at this board?

I've got problmes uploading my code using Platformio because of permanent timeout at correct detected COM.
At other ESP32-boards with same problem I connected GPIO0 to GND and could get upload working again.

How to get access to GPIO0 at LILYGO T3 V2.1.6 or solve this problem by another way?
Pls help...

ianziot, 2021-04-03:
https://github.com/umbm/TTGO-LoRa32-V2.1-T3_V1.6

flux242, 2021-06-17:
i've stumbled upon this issue too. The problem was the esptool version installed on my system. Ubuntu 20.04 provides version 2.8 which doesn't work with this HW. The simplest solution was to keep installed version as it also keeps needed deps and to download only the esptool.py from https://github.com/espressif/esptool. Then it worked without shorting any pin