LilyGO/TTGO-LoRa32-V2.1 #10: free/usable pins

ppisljar, 2021-11-27:
I am trying to figure out if there are any free pins on this development board ? 
looking thru the list published here: https://github.com/Xinyuan-LilyGO/TTGO-LoRa-Series
schematics published here: https://github.com/LilyGO/TTGO-LORA32/blob/master/schematic1in6.pdf
and esp32 pin reservations published here: https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

this is what i came up with: 
does that mean there are no free/usable pins on this board ?

36*                          // input only
39*                          // input only
RST
34*                          // input only
35*                          // input only
14#      sdcard SCLK
12#                          // output only, boot fail if pulled high (in schematics its connected to microSD DAT3?)
13#      sdcard CS
15#      sdcard MOSI
02#      sdcard MISO
00                          // output only, random output at boot
04      <-- only usable pin ? in schematic seems to be connected to microSD DAT1 ?                      
25#     geen led

21      oled SDA
TX
RX
22      oled SCL
23      lora RST
19      lora MISO
Lora1
Lora2
26      lora DIO0
GND
3.3
GND
5v

ppisljar, 2021-11-27:
is maybe pcb design available, i would like to cut away the sdcard part (i need those pins for ADCs and they are currently pulled high)

DonatPlayZ, 2025-01-21:
Hey did you find any Free Pin? I want to connect a buzzer to it. Is the GPIO 04 Useable? i dont use the SD Card